sphero-inc/Sphero-iOS-SDK #21: RKStabilizationCommand and RKMacroObject in SDK 3.0.0-beta2

sodastsai, 2015-02-13:
When playing an ```RKMacroObject``` with ```RKStabilizationCommand``` as one command, the app crashes and raises an exception ```-[RKStabilizationCommand getByteRepresentation]: unrecognized selector sent to instance```.

```objective-c
RKMacroObject *macro = [[RKMacroObject alloc] init];
[macro addCommand:[RKStabilizationCommand commandWithState:RKStabilizationStateOn]];
RKMacroPlayer *player = [[RKMacroPlayer alloc] initWithRobot:self.robot.robot];
[player play:macro];
```
(where ```self.robot``` is an ```RKConvenienceRobot``` instance)

Anyway, it's okay to use 
```objective-c
[self.robot sendCommand:[RKStabilizationCommand commandWithState:RKStabilizationStateOn]];
```

I'm using the new SDK (3.0.0-beta2)

iamcgn, 2015-02-17:
Thanks again for the feedback!  What you're describing works as expected.  RKMacroObject only takes RKMacroCommand objects.  RKStabilizationCommand is a robot control command not an instance of RKMacroCommand.  You currently don't see a compiler warning if you pass the command in inline like your example above because we're returning id rather than instancetype.  I will update the command classes to return instancetype so you get some compile time help in this situation.

sodastsai, 2015-02-19:
Great, it's good to use compiler's features to keep developers away from typo and these type errors. 

Thanks.

zenelk, 2015-02-19:
Added JIRA task for this: SDK-665

zenelk, 2015-03-23:
This is updated in the Beta 3 release

closed 2015-03-23 by zenelk