sphero-inc/Sphero-iOS-SDK #23: Sphero Connectivity

sodastsai, 2015-02-26:
Hi, 

I have a test project [sodastsai/spheroconn](https://github.com/sodastsai/spheroconn) here.
This test project has an issue about Sphero connectivity.

In this app, there's a button for users to connect/disconnect Sphero.

1. At the first time pressing the connect button after launching the app, the Sphero ball is connected and changes to green light. (correct behavior)
2. And then after clicking the **disconnect**, the ball changes white light and the label of the app shows it has been disconnected.
3. The second time pressing the connect button, no thing happens.
4. And at the third time, it works again.

Even the 4th time, 5th time, and further times, the Sphero ball is connectable at N times, and has no response at N+1 times.

weird.

---
Following are my NSLog output of this app (with some comments):
```
2015-02-26 11:58:58.156 SpheroConn[7828:2804657] Button clicked: connect
// First time clicking the button
2015-02-26 11:58:58.160 SpheroConn[7828:2804657] SpheroDevice Created
2015-02-26 11:58:58.257 SpheroConn[7828:2804657] <RKRobotChangedStateNotification  <Sphero @ 6886E700CFD4 [connected:YES connectID:26420000l]> to RKRobotConnecting>
2015-02-26 11:58:58.259 SpheroConn[7828:2804657] <RKRobotChangedStateNotification  <Sphero @ 6886E700CFD4 [connected:YES connectID:26420000l]> to RKRobotOnline>
2015-02-26 11:58:59.436 SpheroConn[7828:2804657] Button clicked: disconnect
2015-02-26 11:58:59.437 SpheroConn[7828:2804657] SpheroDevice Destroyed
// Second time
2015-02-26 11:59:00.420 SpheroConn[7828:2804657] Button clicked: connect
2015-02-26 11:59:00.421 SpheroConn[7828:2804657] SpheroDevice Created
// Hey ... where's the state change notification ...
2015-02-26 11:59:01.569 SpheroConn[7828:2804657] Button clicked: disconnect
2015-02-26 11:59:01.570 SpheroConn[7828:2804657] SpheroDevice Destroyed
// Third time
2015-02-26 11:59:02.767 SpheroConn[7828:2804657] Button clicked: connect
2015-02-26 11:59:02.768 SpheroConn[7828:2804657] SpheroDevice Created
2015-02-26 11:59:02.780 SpheroConn[7828:2804657] <RKRobotChangedStateNotification  <Sphero @ 6886E700CFD4 [connected:YES connectID:26420000l]> to RKRobotConnecting>
2015-02-26 11:59:02.780 SpheroConn[7828:2804657] <RKRobotChangedStateNotification  <Sphero @ 6886E700CFD4 [connected:YES connectID:26420000l]> to RKRobotOnline>
2015-02-26 11:59:03.685 SpheroConn[7828:2804657] Button clicked: disconnect
2015-02-26 11:59:03.685 SpheroConn[7828:2804657] SpheroDevice Destroyed
```

VVes, 2015-02-27:
Test app is broken.  Closing.

closed 2015-02-27 by VVes

sodastsai, 2015-02-27:
Sorry but how's it broken?

VVes, 2015-02-27:
Your retain cycle of the SpheroDevice was incorrect.  I'll post the refactoring if you like.

sodastsai, 2015-02-27:
Okay, I've granted the permission to you

VVes, 2015-02-27:
Pushed the update.