sphero-inc/Sphero-iOS-SDK #25: Fire a RKRobotChangedStateNotification when the sphero ball is put onto the charger.

sodastsai, 2015-03-09:
Currently there's no any events fired when I put a sphero ball to the charger. (but there's a notification with type ```RKRobotOffline``` fired when calling ```-[RKConvenienceRobot sleep]```)

Is it possible to fire a notification for this kind of event? we need this for some UI notifications.
If not, is there any timeout of response to check for this?

zenelk, 2015-03-23:
Thanks for the report, I have this tracked as SDK-751 on our internal project management and we'll get it fixed in the next release.

zenelk, 2015-03-25:
I've just pushed up an update that fixes this. It is a bit slow due to having to rely on the OS telling us that the bluetooth connection has ended, but it does come through now.

closed 2015-03-25 by zenelk