sphero-inc/Sphero-iOS-SDK #28: infinite recursion in RKSession

sodastsai, 2015-03-30:
Hi,

While connecting with Sphero by the 3.0 Beta 3 SDK, sometimes I encounter this crash event about `RKSession`. It seems an infinite recursion. 

```
EXC_BAD_ACCESS KERN_PROTECTION_FAILURE at 0x0000000106ea3ff0

Thread : Crashed: Thread
0   libsystem_malloc.dylib         0x0000000198b323e4 _nano_malloc_check_clear
1   libsystem_malloc.dylib         0x0000000198b31064 nano_malloc + 44
2   libsystem_malloc.dylib         0x0000000198b213e4 malloc_zone_malloc + 116
3   CoreFoundation                 0x00000001876e7318 __CFStringChangeSizeMultiple + 992
4   CoreFoundation                 0x00000001876e5894 __CFStringAppendBytes + 544
5   CoreFoundation                 0x00000001876e0820 __CFStringAppendFormatCore + 7504
6   CoreFoundation                 0x00000001876dea98 _CFStringCreateWithFormatAndArgumentsAux2 + 248
7   Foundation                     0x00000001884f5868 -[NSPlaceholderString initWithFormat:locale:arguments:] + 172
8   Foundation                     0x00000001884f5728 +[NSString stringWithFormat:] + 76
9   SampleApp                      0x0000000100890904 -[RKSession processPartialResponse] (RKSession.m:147)
10  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
11  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
12  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
13  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
14  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
15  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
16  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
17  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
18  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
19  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
20  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
21  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
22  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
23  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
24  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
25  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
26  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
27  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
28  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
29  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
30  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
31  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
32  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
33  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
34  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
35  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
36  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
37  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
38  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
39  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
40  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
41  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
42  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
43  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
44  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
45  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
46  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
47  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
48  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
49  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
50  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
51  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
52  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
53  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
54  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
55  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
56  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
57  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
58  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
59  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
60  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
61  SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
[... 425 lines cut ...]
487 SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
488 SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
489 SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
490 SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
491 SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
492 SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
493 SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
494 SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
495 SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
496 SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
497 SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
498 SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
499 SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
500 SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
501 SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
502 SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
503 SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
[... cut ...]
506 SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
507 SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
508 SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
509 SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
510 SampleApp                      0x0000000100890d50 -[RKSession parseStringResponse] (RKSession.m:195)
```

zenelk, 2015-03-30:
Hey there, I'm having trouble reproducing your issue. Could you tell me which commit you are using? This looks like an issue that was fixed at commit e0fcc6c82625d91c1a97cccc9d41877a10d67cfd on the beta branch.

sodastsai, 2015-03-30:
orbotix/Sphero-iOS-SDK@3609b627381b8ebf135d036fcc11f1ecfb32d000

zenelk, 2015-03-30:
That's the same commit that I'm using. Can you give me a rough idea what you are sending to the robot (and what the robot should be sending back) so that I can try to get it reproducing? 

Also, can you get this problem to occur in any of the samples that are shipped with the SDK frameworks?

zenelk, 2015-03-30:
After taking some time and looking through this, I don't see any way this can happen. I am thinking that you have the wrong version of the framework. In the RKSession class, parseStringRepsonse is never recursively called (in any released versions). Try the newest framework that I just pushed, it should work for you. 

I'm going to leave this issue open for a bit until I'm sure that this is fixed.

sodastsai, 2015-03-31:
Hi, just encounter this again.

This time I met this when switching apps (click the iPad home button quickly).
(as the instruction in demo app, while the app resigning active, the app will disconnect sphero)

At first, I receive following lines in console
```
2015-03-31 18:33:27.989 Tickle[1049:244621] OBX:-[RKSession processPartialResponse] - Correct SOP1, incorrect SOP2. Dumping to string response...
2015-03-31 18:33:27.991 Tickle[1049:244621] OBX:-[RKSession processPartialResponse] - Correct SOP1, incorrect SOP2. Dumping to string response...
2015-03-31 18:33:27.992 Tickle[1049:244621] OBX:-[RKSession processPartialResponse] - Correct SOP1, incorrect SOP2. Dumping to string response...
2015-03-31 18:33:27.993 Tickle[1049:244621] OBX:-[RKSession processPartialResponse] - Correct SOP1, incorrect SOP2. Dumping to string response...
2015-03-31 18:33:27.994 Tickle[1049:244621] OBX:-[RKSession processPartialResponse] - Correct SOP1, incorrect SOP2. Dumping to string response...
2015-03-31 18:33:27.996 Tickle[1049:244621] OBX:-[RKSession processPartialResponse] - Correct SOP1, incorrect SOP2. Dumping to string response...
2015-03-31 18:33:27.997 Tickle[1049:244621] OBX:-[RKSession processPartialResponse] - Correct SOP1, incorrect SOP2. Dumping to string response...
```
(repeated many lines ...)

then crashed and get call traceback like above post.

zenelk, 2015-03-31:
This is with the frameworks pushed in commit d6a47785d06dc71a399b376400f7b45c1ee682af, right? If so, could you give me a snippet of your app will resign active code where you are disconnecting? I still cannot reproduce the issue.

sodastsai, 2015-04-01:
it's d6a4778 right. and the code while receiving `UIApplicationWillResignActiveNotification` is just following lines:
```
- (void)applicationWillResignActive:(NSNotification *)notification {
    [RKRobotDiscoveryAgent stopDiscovery];
    [RKRobotDiscoveryAgent disconnectAll];
}
```

I could not reproduce this issue each time either. Actually it's rare, about 1~2 times during our internal test (about 2 days). I'll update to you if encountering this again.

thalmicMark, 2015-05-06:
I'm noticing this issue as well on `d6a4778`. In our `appWillResignActive` we only call `[RKRobotDiscoveryAgent stopDiscovery]` though. We've got the background bluetooth capability set up, and it looks like Sphero and Ollie can be controlled fine while the app is backgrounded.

zenelk, 2015-05-06:
Hey guys, we have a fix coming down through the pipes and is slated for the next release of the beta SDK. That being said, I actually don't have the timeframe on that release, but I would imagine sooner than later.

sodastsai, 2015-06-03:
[image]
[image]

Any update on this? I've got lots of crash events from users ...

zenelk, 2015-06-03:
We are slating a release for next week, and I haven't seen anything like this in our base for quite some time. Can you check again when that release comes out?

sodastsai, 2015-06-04:
sure. looking forward to it :)

zenelk, 2015-06-12:
Hey sodastsai, our current release didn't quite meet the quality that we wanted, so it had to be delayed. We're still looking for as soon as possible to get it out, and I'll let you know when I find out more.

sodastsai, 2015-06-13:
👍

VVes, 2015-07-07:
Fixed in 3.1

closed 2015-07-07 by VVes