sphero-inc/Sphero-iOS-SDK #31: RKClassicLink crashed

sodastsai, 2015-04-13:
Using SDK @ 8916d9b63e81e0db64bb459b7f3a3de4253426e2
Just report back as a record :)

```
Thread : Crashed: com.orbotis.commandDispatchQueue.8AF8FEC6-347E-42FF-A5AE-6C7A3ACFB672
0  libsystem_pthread.dylib        0x000000019641d318 _pthread_mutex_lock + 40
1  CoreFoundation                 0x00000001848bbf80 CFRunLoopSourceSignal + 64
2  CoreFoundation                 0x00000001848bbf80 CFRunLoopSourceSignal + 64
3  Sample                         0x00000001006036a4 -[RKClassicLink signalSendCommand] (RKClassicLink.m:206)
4  Sample                         0x00000001006034b8 __30-[RKClassicLink queueCommand:]_block_invoke (RKClassicLink.m:184)
5  libdispatch.dylib              0x000000019623d994 _dispatch_call_block_and_release + 24
6  libdispatch.dylib              0x000000019623d954 _dispatch_client_callout + 16
7  libdispatch.dylib              0x00000001962480a4 _dispatch_queue_drain + 1448
8  libdispatch.dylib              0x0000000196240a5c _dispatch_queue_invoke + 132
9  libdispatch.dylib              0x000000019624a318 _dispatch_root_queue_drain + 720
10 libdispatch.dylib              0x000000019624bc4c _dispatch_worker_thread3 + 108
11 libsystem_pthread.dylib        0x000000019641d22c _pthread_wqthread + 816
```

zenelk, 2015-04-13:
Do you have more logs than this or reproduction steps? I can't really gleam what this might be from the small stack trace. Tracked internally as SDK-792

sodastsai, 2015-04-14:
I could not reproduce it yet.
This was also reported by our app's crash logger.
I'll keep update it if I have more information from the crash logger.

iamcgn, 2016-01-06:
We haven't been able to reproduce this issue and we haven't heard any more information about this crash. I'm closing for now. If more information is provided please open a new issue.

closed 2016-01-06 by iamcgn