sphero-inc/Sphero-iOS-SDK #32: -[RKSession popCommandForSequence:]

sodastsai, 2015-04-17:
FYI

we received this crash event in our app's crash logs. 

Currently we have no other information about what did user do before this crash. So we send it back to you as a record. 
I will keep updating it if we have further information. 

```
Thread : Crashed: Thread
0  libobjc.A.dylib                0x301c3f62 objc_msgSend + 1
1  CoreFoundation                 0x21e38f67 -[__NSDictionaryM objectForKey:] + 130
2  Tickle                         0x004f191d -[RKSession popCommandForSequence:] (RKSession.m:236)
3  Tickle                         0x004f1681 -[RKSession processRawData:] (RKSession.m:172)
4  Tickle                         0x004cf5a7 -[RKClassicLink readDataFromStream] (RKClassicLink.m:506)
5  Tickle                         0x004cfac3 -[RKClassicLink stream:handleEvent:] (RKClassicLink.m:591)
6  ExternalAccessory              0x22b80187 -[EAInputStream _streamEventTrigger] + 502
7  CoreFoundation                 0x21ee222f __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 14
8  CoreFoundation                 0x21ee1643 __CFRunLoopDoSources0 + 222
9  CoreFoundation                 0x21edfcc1 __CFRunLoopRun + 768
10 CoreFoundation                 0x21e2b9a1 CFRunLoopRunSpecific + 476
11 CoreFoundation                 0x21e2b7b3 CFRunLoopRunInMode + 106
12 Foundation                     0x22b94dc1 -[NSRunLoop(NSRunLoop) runMode:beforeDate:] + 264
13 Foundation                     0x22bb1bff -[NSRunLoop(NSRunLoop) runUntilDate:] + 86
14 Tickle                         0x004cec43 -[RKClassicLink runCommunicationTask] (RKClassicLink.m:340)
15 Foundation                     0x22c5a18b __NSThread__main__ + 1118
16 libsystem_pthread.dylib        0x308a5deb _pthread_body + 138
17 libsystem_pthread.dylib        0x308a5d5f _pthread_start + 118
```

VVes, 2015-06-25:
Do you have an OS Version and Device version?

sodastsai, 2015-06-25:
iOS 8.3 and iPad 2

iamcgn, 2016-01-06:
@sodastsai are you still seeing this issue?

iamcgn, 2016-02-16:
@sodastsai: We haven't been able to reproduce this issue internally and I haven't heard back on it. I'm closing until we learn more about the problem. If you are still seeing this issue let me know and I will re-open.

closed 2016-02-16 by iamcgn