sphero-inc/Sphero-iOS-SDK #33: `-[RKLeRobotLink isRollCommand:]` crashes

sodastsai, 2015-06-03:
The exception is 
```
Fatal Exception: NSRangeException
*** -[_NSInlineData subdataWithRange:]: range {6, 2} exceeds data length 7
```
and its tracestack is 
```
Thread : Fatal Exception: NSRangeException
0  CoreFoundation                 0x0000000182abc2d8 __exceptionPreprocess
1  libobjc.A.dylib                0x00000001940c80e4 objc_exception_throw
2  CoreFoundation                 0x0000000182abc218 -[NSException initWithCoder:]
3  Foundation                     0x000000018393d738 -[NSData(NSData) subdataWithRange:]
4  Sample                         0x000000010069feb0 -[RKLeRobotLink isRollCommand:] (RKLeRobotLink.m:375)
5  Sample                         0x000000010069e12c -[RKLeRobotLink peripheral:didWriteValueForCharacteristic:error:] (RKLeRobotLink.m:245)
6  CoreBluetooth                  0x0000000182728b3c -[CBPeripheral handleMsg:args:]
7  CoreBluetooth                  0x00000001827249d4 -[CBCentralManager xpcConnection:didReceiveMsg:args:]
8  libdispatch.dylib              0x0000000194719994 _dispatch_call_block_and_release
9  libdispatch.dylib              0x0000000194719954 _dispatch_client_callout
10 libdispatch.dylib              0x00000001947240a4 _dispatch_queue_drain
11 libdispatch.dylib              0x000000019471ca5c _dispatch_queue_invoke
12 libdispatch.dylib              0x000000019471ddbc _dispatch_main_queue_callback_4CF
13 CoreFoundation                 0x0000000182a737f8 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__
14 CoreFoundation                 0x0000000182a718a0 __CFRunLoopRun
15 CoreFoundation                 0x000000018299d2d4 CFRunLoopRunSpecific
16 GraphicsServices               0x000000018c0736fc GSEventRunModal
17 UIKit                          0x0000000187562fac UIApplicationMain
18 Sample                         0x00000001000d2b40 main (main.m:14)
19 libdyld.dylib                  0x0000000194746a08 start
```

VVes, 2015-06-25:
Please update to SDK 3.1.14 .

sodastsai, 2015-06-25:
Hey, is it possible to add a git tag for 3.1.14 release? thx.

VVes, 2015-06-25:
Done.  It's current Master.

closed 2015-06-25 by VVes