sphero-inc/Sphero-iOS-SDK #34: `-[RKLeRobotLink sendCommand:streaming:]` crashes

sodastsai, 2015-06-03:
FYI.

The exception is:
```
Fatal Exception: NSInvalidArgumentException
*** -[__NSPlaceholderDictionary initWithObjects:forKeys:count:]: attempt to insert nil object from objects[1]
```
And traceback info is:
```
Thread : Fatal Exception: NSInvalidArgumentException
0  CoreFoundation                 0x23482fef __exceptionPreprocess + 126
1  libobjc.A.dylib                0x3186ec8b objc_exception_throw + 38
2  CoreFoundation                 0x233a16af -[__NSPlaceholderDictionary initWithObjects:forKeys:count:] + 502
3  CoreFoundation                 0x233a1493 +[NSDictionary dictionaryWithObjects:forKeys:count:] + 50
4  Sample                         0x00610feb __39-[RKLeRobotLink sendCommand:streaming:]_block_invoke (RKLeRobotLink.m:423)
5  libdispatch.dylib              0x31dd92e3 _dispatch_call_block_and_release + 10
6  libdispatch.dylib              0x31de1729 _dispatch_queue_drain + 1468
7  libdispatch.dylib              0x31ddbaad _dispatch_queue_invoke + 84
8  libdispatch.dylib              0x31de2f9f _dispatch_root_queue_drain + 394
9  libdispatch.dylib              0x31de43c3 _dispatch_worker_thread3 + 94
10 libsystem_pthread.dylib        0x31f40dc1 _pthread_wqthread + 668
11 libsystem_pthread.dylib        0x31f40b14 start_wqthread + 8
```
on queue `com.orbotix.commandDispatchQueue.0x1b079d30`

sodastsai, 2015-08-17:
Another issue occurred in this method is 

```
Thread : Crashed: com.apple.root.user-initiated-qos
0  libobjc.A.dylib                0x313f8f46 objc_msgSend + 5 (objectForKeyedSubscript:)
1  Tickle                         0x00599c4b -[RKLeRobotLink sendCommand:streaming:] (RKLeRobotLink.m:390)
2  Tickle                         0x00599b37 -[RKLeRobotLink sendCommand:] (RKLeRobotLink.m:382)
3  Tickle                         0x005a7e2d -[RKRobotLE sendCommand:] (RKRobotLE.m:198)
4  Tickle                         0x00540a9d -[RKConvenienceRobot setBackLEDBrightness:] (RKConvenienceRobot.m:82)
5  Sample                         0x003a4e6f __39-
6  libdispatch.dylib              0x319572e3 _dispatch_call_block_and_release + 10
7  libdispatch.dylib              0x31961381 _dispatch_root_queue_drain + 1388
8  libdispatch.dylib              0x319623c3 _dispatch_worker_thread3 + 94
9  libsystem_pthread.dylib        0x31abedc1 _pthread_wqthread + 668
10 libsystem_pthread.dylib        0x31abeb14 start_wqthread + 8
```

iamcgn, 2016-01-06:
@sodastsai are you still seeing this issue?

iamcgn, 2016-02-16:
@sodastsai: this issue should have been fixed in the last release. We haven't been able to reproduce it internally. If you're still seeing the crash let me know and I will re-open this issue.

closed 2016-02-16 by iamcgn