sphero-inc/Sphero-iOS-SDK #35: RKStatRecorder issues

sodastsai, 2015-06-04:
FYI. I've got 2 kind of exceptions related to `RKStatRecorder` from my crash report collector.

The first is in `-[RKStatRecorder createNewRandomFileHandleFromPath:]`, `RKStatRecorder.m line 285`
```
Fatal Exception: Primary stats file failed
Cannot create primary stats file at /var/mobile/Containers/Data/Application/0D71A38B-44A1-4AB5-A2AE-317565C4C00D/Documents/2582DC44-E949-42A2-A4A8-FF14B1B9EA00.statcache
```
and traceback is
```
Thread : Fatal Exception: Primary stats file failed
0  CoreFoundation                 0x25fb8137 __exceptionPreprocess + 126
1  libobjc.A.dylib                0x34253c77 objc_exception_throw + 38
2  CoreFoundation                 0x25fb807d -[NSException initWithCoder:]
3  Sample                         0x005d3843 -[RKStatRecorder createNewRandomFileHandleFromPath:] (RKStatRecorder.m:285)
4  Sample                         0x005d32af -[RKStatRecorder serializeAndWriteStatsToFile:] (RKStatRecorder.m:254)
5  Sample                         0x005d3193 -[RKStatRecorder writeStatsToFiles:] (RKStatRecorder.m:246)
6  Sample                         0x005d258f __34-[RKStatRecorder syncToPersistent]_block_invoke (RKStatRecorder.m:166)
7  libdispatch.dylib              0x347bdc83 _dispatch_call_block_and_release + 10
8  libdispatch.dylib              0x347c84d3 _dispatch_queue_drain$VARIANT$mp + 1462
9  libdispatch.dylib              0x347c7da1 _dispatch_queue_invoke$VARIANT$mp + 84
10 libdispatch.dylib              0x347ca491 _dispatch_root_queue_drain + 388
11 libdispatch.dylib              0x347cb8a3 _dispatch_worker_thread3 + 106
12 libsystem_pthread.dylib        0x3493fda9 _pthread_wqthread + 668
13 libsystem_pthread.dylib        0x3493fafc start_wqthread + 8
```
This only happens to only 1 user but with a high frequency. I guess it's maybe due to this user's iPad is full and doesn't have enough space.

---

The second is in `-[RKStatRecorder syncToPersistent]`, `RKStatRecorder.m line 162`
```
Fatal Exception: NSInvalidArgumentException
*** -[__NSPlaceholderArray initWithObjects:count:]: attempt to insert nil object from objects[28]
```
and traceback is 
```
Thread : Fatal Exception: NSInvalidArgumentException
0  CoreFoundation                 0x24e6a137 __exceptionPreprocess + 126
1  libobjc.A.dylib                0x33105c77 objc_exception_throw + 38
2  CoreFoundation                 0x24d836b1 -[__NSPlaceholderArray initWithObjects:count:] + 428
3  CoreFoundation                 0x24d83f1f -[NSArray initWithArray:range:copyItems:] + 274
4  CoreFoundation                 0x24d83df5 +[NSArray arrayWithArray:] + 72
5  Sample                         0x005139e3 __34-[RKStatRecorder syncToPersistent]_block_invoke (RKStatRecorder.m:162)
6  libdispatch.dylib              0x3366fc83 _dispatch_call_block_and_release + 10
7  libdispatch.dylib              0x3367a4d3 _dispatch_queue_drain$VARIANT$mp + 1462
8  libdispatch.dylib              0x33679da1 _dispatch_queue_invoke$VARIANT$mp + 84
9  libdispatch.dylib              0x3367c491 _dispatch_root_queue_drain + 388
10 libdispatch.dylib              0x3367d8a3 _dispatch_worker_thread3 + 106
11 libsystem_pthread.dylib        0x337f1da9 _pthread_wqthread + 668
12 libsystem_pthread.dylib        0x337f1afc start_wqthread + 8
```
This crash is not a high frequency issue. The number `28` (objects[28]) in the exception description is not always 28. It's sometimes 0, 140, or other numbers.

VVes, 2015-07-07:
Opened an internal change request to disable robot stats when mobile device is out of space.

iamcgn, 2016-10-20:
This issue is fixed in the latest beta branch build.

closed 2016-10-20 by iamcgn

VVes, 2016-10-20:
Thank goodness. :)