sphero-inc/Sphero-iOS-SDK #36: [CBPeripheral UUID]: unrecognized selector sent to instance

thalmicMark, 2015-06-12:
Not sure if you guys are looking at supporting iOS 9 yet, but we've seen this bug appear. Should be able to replace all uses of `UUID` with the equivalent `identifier`.

Environment:
version: iOS 9 beta 1
device: iPhone 6

Stack trace: 

    Thread : Fatal Exception: NSInvalidArgumentException
    0  ???                            0x000000018426ff44 
    1  ???                            0x000000019928bf2c 
    2  ???                            0x0000000184276c3c 
    3  ???                            0x0000000184273be0 
    4  ???                            0x0000000184177b2c 
    5  MyoSphero                      0x00000001001726b4 -[RKRobotLE identifier] (RKRobotLE.m:108)
    6  MyoSphero                      0x000000010014e754 -[RKStatEventListener handleRobotChangedState:] (RKStatEventListener.m:68)
    7  ???                            0x0000000184217120 
    8  ???                            0x0000000184216940 
    9  ???                            0x00000001842166cc 
    10 ???                            0x00000001842781d0 
    11 ???                            0x0000000184152b24 
    12 MyoSphero                      0x0000000100170cc4 -[RKRobotDiscoveryNotificationRelay nodeStateDidChange:] (RKRobotDiscoveryAgentLE.m:36)
    13 ???                            0x0000000184275a50 
    14 ???                            0x0000000184173398 
    15 MyoSphero                      0x000000010016c9c0 -[RKDiscoveryAgentLE postStateChangedNotification:] (RKDiscoveryAgentLE.m:346)
    16 MyoSphero                      0x000000010016c1e8 -[RKDiscoveryAgentLE nodeDidWake:] (RKDiscoveryAgentLE.m:311)
    17 MyoSphero                      0x00000001001736d0 -[RKRobotLE didWake] (RKRobotLE.m:236)
    18 MyoSphero                      0x0000000100165aa8 -[RKLeRobotLink postOnline] (RKLeRobotLink.m:575)
    19 MyoSphero                      0x00000001001631f8 -[RKLeRobotLink handleSpecialResponses:] (RKLeRobotLink.m:337)
    20 MyoSphero                      0x0000000100173ddc -[RKRobotLE handleResponse:] (RKRobotLE.m:294)
    21 MyoSphero                      0x0000000100158eac -[RKSession parseResponse] (RKSession.m:163)
    22 MyoSphero                      0x0000000100158bf0 -[RKSession processPartialResponse] (RKSession.m:141)
    23 MyoSphero                      0x0000000100158ab0 -[RKSession processRawData:] (RKSession.m:125)
    24 MyoSphero                      0x0000000100162b08 -[RKLeRobotLink peripheral:didUpdateValueForCharacteristic:error:] (RKLeRobotLink.m:307)
    25 ???                            0x0000000183ec8adc 
    26 ???                            0x0000000183ed445c 
    27 ???                            0x0000000199a95cdc 
    28 ???                            0x0000000199a95c9c 
    29 ???                            0x0000000199a9ae50 
    30 ???                            0x0000000184227ce0 
    31 ???                            0x0000000184225b1c 
    32 ???                            0x00000001841510c0 
    33 ???                            0x000000018f4cf170 
    34 ???                            0x00000001897ae5b8 
    35 MyoSphero                      0x00000001000ab22c main (main.m:16)
    36 ???                            0x0000000199ac69e8

VVes, 2015-06-25:
We will have iOS9 support a couple weeks before it's official release.

VVes, 2015-07-07:
We have officially started on iOS9 support and will be releasing an update by 7/17.

thalmicMark, 2015-07-07:
Cool thanks for the update!

thalmicMark, 2015-08-14:
What's the status of this? We'd like to push an update to our app before iOS 9 drops.

VVes, 2015-08-17:
We have been using a stable iOS9 Sphero SDK internally the last couple weeks and will be releasing it publicly this week.  (Week of Aug 17th.)

profmike, 2015-08-24:
any update on the release?

VVes, 2015-08-24:
It will drop tomorrow.

iamcgn, 2015-08-24:
The new release is up with iOS 9 compatibility fixes!

closed 2015-08-24 by iamcgn