sphero-inc/Sphero-iOS-SDK #38: -[RKDeviceSensorsData initWithMask:data:] crash

sodastsai, 2015-08-20:
FYI. Using the latest SDK 3.1.14

```
Thread : Fatal Exception: NSRangeException
0  CoreFoundation                 0x000000018544c2d8 __exceptionPreprocess
1  libobjc.A.dylib                0x00000001968f80e4 objc_exception_throw
2  CoreFoundation                 0x000000018544c218 -[NSException initWithCoder:]
3  Foundation                     0x00000001862804d4 -[NSData(NSData) getBytes:range:]
4  Sample                         0x0000000100757b74 -[RKDeviceSensorsData initWithMask:data:] (RKDeviceSensorsData.m:194)
5  Sample                         0x0000000100763594 -[RKDeviceSensorsAsyncData initWithAsyncPacket:] (RKDeviceSensorsAsyncData.m:66)
6  Sample                         0x000000010075fc24 +[RKAsyncFactory dataFromPacket:] (RKAsyncFactory.m:65)
7  Sample                         0x00000001007678d0 -[RKSession processRawData:] (RKSession.m:190)
8  Sample                         0x000000010072a550 -[RKClassicLink readDataFromStream] (RKClassicLink.m:556)
9  Sample                         0x000000010072ac4c -[RKClassicLink stream:handleEvent:] (RKClassicLink.m:649)
10 ExternalAccessory              0x000000018624c71c -[EAInputStream _streamEventTrigger]
11 CoreFoundation                 0x0000000185404240 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__
12 CoreFoundation                 0x00000001854034e4 __CFRunLoopDoSources0
13 CoreFoundation                 0x0000000185401594 __CFRunLoopRun
14 CoreFoundation                 0x000000018532d2d4 CFRunLoopRunSpecific
15 Foundation                     0x00000001862654c8 -[NSRunLoop(NSRunLoop) runMode:beforeDate:]
16 Foundation                     0x0000000186287a60 -[NSRunLoop(NSRunLoop) runUntilDate:]
17 Sample                         0x0000000100729a48 -[RKClassicLink runCommunicationTask] (RKClassicLink.m:381)
18 Foundation                     0x000000018634ddb8 __NSThread__main__
19 libsystem_pthread.dylib        0x000000019712bdc8 _pthread_body
20 libsystem_pthread.dylib        0x000000019712bd24 _pthread_body
21 libsystem_pthread.dylib        0x0000000197128ef8 thread_start
```

PaulTR, 2016-01-19:
We haven't been able to reproduce this in the current SDK. Is this still an issue for you? If so, what mask are you using when requesting the data?

PaulTR, 2016-01-30:
Could not reproduce, and didn't hear back on the issue after ten days, so closing.

closed 2016-01-30 by PaulTR