sphero-inc/Sphero-iOS-SDK #39: Push Notification Entitlements

thalmicMark, 2015-08-27:
We received this warning from Apple when submitting our latest release containing Sphero SDK 3.1.78:

> Missing Push Notification Entitlement - Your app appears to include API used to register with the Apple Push Notification service, but the app signature's entitlements do not include the "aps-environment" entitlement. If your app uses the Apple Push Notification service, make sure your App ID is enabled for Push Notification in the Provisioning Portal, and resubmit after signing your app with a Distribution provisioning profile that includes the "aps-environment" entitlement. See "Provisioning and Development" in the Local and Push Notification Programming Guide for more information. If your app does not use the Apple Push Notification service, no action is required. You may remove the API from future submissions to stop this warning. If you use a third-party framework, you may need to contact the developer for information on removing the API.

zenelk, 2015-08-31:
I ran through all of our iOS library code and verified with the other developers that we are not using any sort of push APIs in our libraries. Is it possible that another library may have brought this warning on? I have been trying to see if perhaps another API may be confused by the Apple static analyzer as a push API, but to no avail.

danbalzerson, 2015-08-31:
With further investigation (I work with Mark), it seems like this issue is probably being caused by Fabric/Crashlytics tripping Apple's static analysis (https://forums.developer.apple.com/thread/11704).  Doesn't seem like its coming from Sphero.

zenelk, 2015-08-31:
Okay, I'm going to go ahead and close this issue then.

closed 2015-08-31 by zenelk