sphero-inc/Sphero-iOS-SDK #40: I can not connect and control more than one Sphero at a time.

michaeldeitcher, 2015-10-11:
The SDK is written like this should just work. When I try to do this with 2 Spheros, I get two RKRobotConnected notifications but only one RKRobotOnline notification. 

I'd really love to know when you will enable this feature. It is vital to the project I'm working on. Thanks

PaulTR, 2016-01-19:
Looks like this works fine. I tested it in the ButtonDrive sample application with the following updated ViewController

```objective-c
@interface ButtonDriveViewController()

@property (strong, nonatomic) RKConvenienceRobot* robot;
@property (strong, nonatomic) RKConvenienceRobot* robot2;
@property (strong, nonatomic) RUICalibrateGestureHandler *calibrateHandler;

@end

@implementation ButtonDriveViewController

- (void)viewDidLoad {
	[super viewDidLoad];
	
	[[NSNotificationCenter defaultCenter] addObserver:self
											 selector:@selector(appWillResignActive:)
												 name:UIApplicationWillResignActiveNotification
											   object:nil];
	[[NSNotificationCenter defaultCenter] addObserver:self
											 selector:@selector(appDidBecomeActive:)
												 name:UIApplicationDidBecomeActiveNotification
											   object:nil];

	
	self.calibrateHandler = [[RUICalibrateGestureHandler alloc] initWithView:self.view];

    [RKRobotDiscoveryAgent sharedAgent].maxConnectedRobots = 2;

	// hook up for robot state changes
	[[RKRobotDiscoveryAgent sharedAgent] addNotificationObserver:self selector:@selector(handleRobotStateChangeNotification:)];

}

- (void)appDidBecomeActive:(NSNotification*)notification {
	[RKRobotDiscoveryAgent startDiscovery];
}

- (void)appWillResignActive:(NSNotification*)notification {
	[RKRobotDiscoveryAgent stopDiscovery];
    [RKRobotDiscoveryAgent disconnectAll];
}

- (void)handleRobotStateChangeNotification:(RKRobotChangedStateNotification*)n {
    switch(n.type) {
        case RKRobotConnecting:
            [self handleConnecting];
            break;
        case RKRobotOnline: {
            // Do not allow the robot to connect if the application is not running)
            if( self.robot == nil ) {
                self.robot = [RKConvenienceRobot convenienceWithRobot:n.robot];
                [self.robot2 setLEDWithRed:0 green:255 blue:255];
                if ([[UIApplication sharedApplication] applicationState] != UIApplicationStateActive) {
                    [self.robot disconnect];
                    return;
                }
            } else {
                self.robot2 = [RKConvenienceRobot convenienceWithRobot:n.robot];
                [self.robot2 setLEDWithRed:255 green:255 blue:0];
                if ([[UIApplication sharedApplication] applicationState] != UIApplicationStateActive) {
                    [self.robot2 disconnect];
                    return;
                }
            }
            [self handleConnected];
            break;
        }
        case RKRobotDisconnected:
            [self handleDisconnected];
            self.robot = nil;
            [RKRobotDiscoveryAgent startDiscovery];
            break;
        default:
            break;
    }
}

- (void)handleConnecting {
    // Handle when a robot is connecting here
}

- (void)handleConnected {
    [_calibrateHandler setRobot:_robot.robot];
}

- (void)handleDisconnected {
    [_calibrateHandler setRobot:nil];
}

- (IBAction)zeroPressed:(id)sender {
	[_robot driveWithHeading:0.0 andVelocity:0.5];
    [_robot2 driveWithHeading:0.0 andVelocity:0.5];
}

- (IBAction)ninetyPressed:(id)sender {
	[_robot driveWithHeading:90 andVelocity:0.5];
	[_robot2 driveWithHeading:90 andVelocity:0.5];
}

- (IBAction)oneEightyPressed:(id)sender {
	[_robot driveWithHeading:180 andVelocity:0.5];
	[_robot2 driveWithHeading:180 andVelocity:0.5];
}

- (IBAction)twoSeventyPressed:(id)sender {
	[_robot driveWithHeading:270.0 andVelocity:0.5];
	[_robot2 driveWithHeading:270.0 andVelocity:0.5];
}

- (IBAction)stopPressed:(id)sender {
	[_robot stop];
	[_robot2 stop];
//    [RKRobotDiscoveryAgent disconnectAll];
}

@end
```

closed 2016-01-19 by PaulTR

michaeldeitcher, 2016-02-06:
Thanks so much. I wasn't setting the maxConnectedRobots value in the discovery agent. Your example code works like a charm. :)