sphero-inc/Sphero-iOS-SDK #43: kRobotRemoteRSSIChangedNotification seems not working.

citysite102, 2015-12-14:
Hello, everything works perfect but the kRobotRemoteRSSIChangedNotification seems not working after connecting to a Ollie device, is there any command need to be sent to Ollie for initialization?

citysite102, 2015-12-14:
And there is another issue about the connection of multiple sphero under iOS SDK, sometimes the Sphero device will stop sending any robot state notification although the [[RKDiscoveryAgentClassic sharedAgent] isDiscovering] = YES. and we cannot find the Sphero in the available robots of [RKDiscoveryAgentClassic sharedAgent] although we already connect to multiple sphero from the bluetooth setting page.

iamcgn, 2016-01-06:
@citysite102 Remote RSSI isn't a supported command on Ollie.  I apologize that this hasn't been communicated through the SDK.

closed 2016-01-06 by iamcgn