sphero-inc/Sphero-iOS-SDK #44: Unable to archive with Xcode

TarangKhanna, 2016-02-14:
My code builds and runs but when i try to archive it the RobotKit(RKOllie.o) causes problems which I get from the Sphero iOS SDK. The error is:
ld: bitcode bundle could not be generated because '/Users/THacked96/Downloads/Sphero-iOS-SDK-master/frameworks/RobotKit.framework/RobotKit(RKOllie.o)' was built without full bitcode. All object files and libraries for bitcode must be generated from Xcode Archive or Install build for architecture armv7
clang: error: linker command failed with exit code 1 (use -v to see invocation)

I tried adding -fembed-bitcode to Other C Flags but that didn't help.

iamcgn, 2016-02-16:
Hi @TarangKhanna,

I'm looking into this issue. In the mean time you can disable bitcode in the main project and continue to use this framework. The -fembed-bitcode flag only works if you're building the framework from source. That flag tells the compiler to add the bitcode section in the binary.  We did build the frameworks with this flag, it just looks like a debug vs release build issue.  I will update you when I have more info.

TarangKhanna, 2016-02-16:
thanks! I have another problem target ' Extension' has bitcode disabled (ENABLE_BITCODE = NO), but it is required for the 'watchos' platform, this shouldn't happen because I am only running the sphero library with the iOS target

iamcgn, 2016-02-17:
It sounds like you turned bitcode off at the project level rather than the target level. If you click on the iOS app target you can disable bitcode for that target only.

pip8786, 2016-09-21:
Using the command line tools, I had to remove the Info.plist from the RobotKitUI.bundle in order to be able to use xcodebuild -exportArchive to create an IPA. Not sure if it's related to this issue.