sphero-inc/Sphero-iOS-SDK #47: RKMagnetometerData

citysite102, 2016-03-12:
Hello, I try to get the Magnetometer sensor data from the device with following data streaming mask:
 `RKDataStreamingMask mask =  RKDataStreamingMaskAccelerometerFilteredAll |
        RKDataStreamingMaskGyroXFiltered |
        RKDataStreamingMaskGyroYFiltered |
        RKDataStreamingMaskGyroZFiltered |
        RKDataStreamingMaskMagnetometerZFiltered |
        RKDataStreamingMaskIMUAnglesFilteredAll;`
But I'll get an error as following.
[image]
with the error message like
 `*** Terminating app due to uncaught exception 'NSRangeException', reason: '*** -[_NSInlineData getBytes:range:]: range {20, 1} exceeds data length 20' `
However, if I remove RKDataStreamingMaskMagnetometerZFiltered from the mask, everything just works perfect : ], is there any information or restriction about the Magnetometer sensor data?

zenelk, 2016-10-05:
I'll get this on the queue to be looked at. I can't think of anything for you to try in the meantime, however.

zenelk, 2016-11-03:
Hey there, so in reviewing the stack, the reason this is occurring is that magnetometer hasn't actually been present in robots since the older versions of Sphero. This crash is happening because you're requesting data that doesn't exist in the packet. I can get this on the board to have a bounds check, but my guess is that it will be a really low priority. For now you should just not request the data.

closed 2016-11-03 by zenelk