sphero-inc/Sphero-iOS-SDK #50: Module Debugging Enabled in Build?

pip8786, 2016-09-21:
It seems that the latest builds have module debugging enabled and references a bunch of PCM files that are not available. Log below:

```
GenerateDSYMFile /Users/phsabo/Library/Developer/Xcode/DerivedData/SpheroBlockly-fswnylmsnxzaxmcrxihrrxyofath/Build/Products/Debug-iphoneos/SpheroBlockly.app.dSYM /Users/phsabo/Library/Developer/Xcode/DerivedData/SpheroBlockly-fswnylmsnxzaxmcrxihrrxyofath/Build/Products/Debug-iphoneos/SpheroBlockly.app/SpheroBlockly
    cd /Users/phsabo/Documents/SAS_Projects/MobilePathways/CodeSnaps_iOS/SpheroBlockly-iOS
    export PATH="/Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/usr/bin:/Applications/Xcode.app/Contents/Developer/usr/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin"
    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/dsymutil /Users/phsabo/Library/Developer/Xcode/DerivedData/SpheroBlockly-fswnylmsnxzaxmcrxihrrxyofath/Build/Products/Debug-iphoneos/SpheroBlockly.app/SpheroBlockly -o /Users/phsabo/Library/Developer/Xcode/DerivedData/SpheroBlockly-fswnylmsnxzaxmcrxihrrxyofath/Build/Products/Debug-iphoneos/SpheroBlockly.app.dSYM

while processing /Users/phsabo/Documents/SAS_Projects/MobilePathways/CodeSnaps_iOS/SpheroBlockly-iOS/RobotKit/RobotKit.framework/RobotKit(RKOllie.o):
warning: /Users/Shared/TeamCity/bamboo-home/xml-data/build-dir/SDK-SDKI-ISR/build/ModuleCache/1W1M0MFNPGILX/Foundation-A3SOD99KJ0S9.pcm: No such file or directory
note: Linking a static library that was built with -gmodules, but the module cache was not found.  Redistributable static libraries should never be built with module debugging enabled.  The debug experience will be degraded due to incomplete debug information.
while processing /Users/phsabo/Documents/SAS_Projects/MobilePathways/CodeSnaps_iOS/SpheroBlockly-iOS/RobotKit/RobotKit.framework/RobotKit(RKOllie.o):
warning: /Users/Shared/TeamCity/bamboo-home/xml-data/build-dir/SDK-SDKI-ISR/build/ModuleCache/1W1M0MFNPGILX/UIKit-2LM3EQU7VVY4O.pcm: No such file or directory
while processing /Users/phsabo/Documents/SAS_Projects/MobilePathways/CodeSnaps_iOS/SpheroBlockly-iOS/RobotKit/RobotKit.framework/RobotKit(RKOllie.o):
warning: /Users/Shared/TeamCity/bamboo-home/xml-data/build-dir/SDK-SDKI-ISR/build/ModuleCache/1W1M0MFNPGILX/CoreGraphics-1YQ59ILDR3NYI.pcm: No such file or directory
while processing /Users/phsabo/Documents/SAS_Projects/MobilePathways/CodeSnaps_iOS/SpheroBlockly-iOS/RobotKit/RobotKit.framework/RobotKit(RKOllie.o):
warning: /Users/Shared/TeamCity/bamboo-home/xml-data/build-dir/SDK-SDKI-ISR/build/ModuleCache/1W1M0MFNPGILX/CoreBluetooth-OTHDZYNRH8AI.pcm: No such file or directory
while processing /Users/phsabo/Documents/SAS_Projects/MobilePathways/CodeSnaps_iOS/SpheroBlockly-iOS/RobotKit/RobotKit.framework/RobotKit(RKOllie.o):
warning: /Users/Shared/TeamCity/bamboo-home/xml-data/build-dir/SDK-SDKI-ISR/build/ModuleCache/1W1M0MFNPGILX/ObjectiveC-2CD0WTQVXITO7.pcm: No such file or directory
while processing /Users/phsabo/Documents/SAS_Projects/MobilePathways/CodeSnaps_iOS/SpheroBlockly-iOS/RobotKit/RobotKit.framework/RobotKit(RKRobotDiscoveryAgent.o):
warning: /Users/Shared/TeamCity/bamboo-home/xml-data/build-dir/SDK-SDKI-ISR/build/ModuleCache/1W1M0MFNPGILX/ExternalAccessory-1YI0BV7LDRAQC.pcm: No such file or directory
while processing /Users/phsabo/Documents/SAS_Projects/MobilePathways/CodeSnaps_iOS/SpheroBlockly-iOS/RobotKit/RobotKit.framework/RobotKit(RKRobotDiscoveryAgent.o):
warning: /Users/Shared/TeamCity/bamboo-home/xml-data/build-dir/SDK-SDKI-ISR/build/ModuleCache/1W1M0MFNPGILX/CoreMotion-RZ4DJFEI29KM.pcm: No such file or directory
while processing /Users/phsabo/Documents/SAS_Projects/MobilePathways/CodeSnaps_iOS/SpheroBlockly-iOS/RobotKit/RobotKit.framework/RobotKit(RKRobotDiscoveryAgent.o):
warning: /Users/Shared/TeamCity/bamboo-home/xml-data/build-dir/SDK-SDKI-ISR/build/ModuleCache/1W1M0MFNPGILX/CFNetwork-929P33Y1WR7E.pcm: No such file or directory
while processing /Users/phsabo/Documents/SAS_Projects/MobilePathways/CodeSnaps_iOS/SpheroBlockly-iOS/RobotKit/RobotKit.framework/RobotKit(RKRobotDiscoveryAgent.o):
warning: /Users/Shared/TeamCity/bamboo-home/xml-data/build-dir/SDK-SDKI-ISR/build/ModuleCache/1W1M0MFNPGILX/Darwin-MI6WZSG1PNOM.pcm: No such file or directory
while processing /Users/phsabo/Documents/SAS_Projects/MobilePathways/CodeSnaps_iOS/SpheroBlockly-iOS/RobotKit/RobotKit.framework/RobotKit(RKClassicLink.o):
warning: /Users/Shared/TeamCity/bamboo-home/xml-data/build-dir/SDK-SDKI-ISR/build/ModuleCache/1W1M0MFNPGILX/CoreFoundation-10I2D2XL7L7X9.pcm: No such file or directory
while processing /Users/phsabo/Documents/SAS_Projects/MobilePathways/CodeSnaps_iOS/SpheroBlockly-iOS/RobotKit/RobotKit.framework/RobotKit(mock_spheroovmops.o):
warning: /Users/Shared/TeamCity/bamboo-home/xml-data/build-dir/SDK-SDKI-ISR/build/ModuleCache/U35GI1AGQ3QQ/Darwin-MI6WZSG1PNOM.pcm: No such file or directory
while processing /Users/phsabo/Documents/SAS_Projects/MobilePathways/CodeSnaps_iOS/SpheroBlockly-iOS/RobotKit/RobotUIKit.framework/RobotUIKit(RUICalibrateOverlayView.o):
warning: /Users/Shared/TeamCity/bamboo-home/xml-data/build-dir/SDK-SDKI-ISR/build/ModuleCache/1W1M0MFNPGILX/QuartzCore-3I7S0M4UEAD0U.pcm: No such file or directory
while processing /Users/phsabo/Documents/SAS_Projects/MobilePathways/CodeSnaps_iOS/SpheroBlockly-iOS/RobotKit/RobotKit.framework/RobotKit(RKOllie.o):
warning: /Users/Shared/TeamCity/bamboo-home/xml-data/build-dir/SDK-SDKI-ISR/build/ModuleCache/FBAX60G1D2VV/Foundation-A3SOD99KJ0S9.pcm: No such file or directory
note: Linking a static library that was built with -gmodules, but the module cache was not found.  Redistributable static libraries should never be built with module debugging enabled.  The debug experience will be degraded due to incomplete debug information.
while processing /Users/phsabo/Documents/SAS_Projects/MobilePathways/CodeSnaps_iOS/SpheroBlockly-iOS/RobotKit/RobotKit.framework/RobotKit(RKOllie.o):
warning: /Users/Shared/TeamCity/bamboo-home/xml-data/build-dir/SDK-SDKI-ISR/build/ModuleCache/FBAX60G1D2VV/UIKit-2LM3EQU7VVY4O.pcm: No such file or directory
while processing /Users/phsabo/Documents/SAS_Projects/MobilePathways/CodeSnaps_iOS/SpheroBlockly-iOS/RobotKit/RobotKit.framework/RobotKit(RKOllie.o):
warning: /Users/Shared/TeamCity/bamboo-home/xml-data/build-dir/SDK-SDKI-ISR/build/ModuleCache/FBAX60G1D2VV/CoreGraphics-1YQ59ILDR3NYI.pcm: No such file or directory
while processing /Users/phsabo/Documents/SAS_Projects/MobilePathways/CodeSnaps_iOS/SpheroBlockly-iOS/RobotKit/RobotKit.framework/RobotKit(RKOllie.o):
warning: /Users/Shared/TeamCity/bamboo-home/xml-data/build-dir/SDK-SDKI-ISR/build/ModuleCache/FBAX60G1D2VV/CoreBluetooth-OTHDZYNRH8AI.pcm: No such file or directory
while processing /Users/phsabo/Documents/SAS_Projects/MobilePathways/CodeSnaps_iOS/SpheroBlockly-iOS/RobotKit/RobotKit.framework/RobotKit(RKOllie.o):
warning: /Users/Shared/TeamCity/bamboo-home/xml-data/build-dir/SDK-SDKI-ISR/build/ModuleCache/FBAX60G1D2VV/ObjectiveC-2CD0WTQVXITO7.pcm: No such file or directory
while processing /Users/phsabo/Documents/SAS_Projects/MobilePathways/CodeSnaps_iOS/SpheroBlockly-iOS/RobotKit/RobotKit.framework/RobotKit(RKRobotDiscoveryAgent.o):
warning: /Users/Shared/TeamCity/bamboo-home/xml-data/build-dir/SDK-SDKI-ISR/build/ModuleCache/FBAX60G1D2VV/ExternalAccessory-1YI0BV7LDRAQC.pcm: No such file or directory
while processing /Users/phsabo/Documents/SAS_Projects/MobilePathways/CodeSnaps_iOS/SpheroBlockly-iOS/RobotKit/RobotKit.framework/RobotKit(RKRobotDiscoveryAgent.o):
warning: /Users/Shared/TeamCity/bamboo-home/xml-data/build-dir/SDK-SDKI-ISR/build/ModuleCache/FBAX60G1D2VV/CoreMotion-RZ4DJFEI29KM.pcm: No such file or directory
while processing /Users/phsabo/Documents/SAS_Projects/MobilePathways/CodeSnaps_iOS/SpheroBlockly-iOS/RobotKit/RobotKit.framework/RobotKit(RKRobotDiscoveryAgent.o):
warning: /Users/Shared/TeamCity/bamboo-home/xml-data/build-dir/SDK-SDKI-ISR/build/ModuleCache/FBAX60G1D2VV/CFNetwork-929P33Y1WR7E.pcm: No such file or directory
while processing /Users/phsabo/Documents/SAS_Projects/MobilePathways/CodeSnaps_iOS/SpheroBlockly-iOS/RobotKit/RobotKit.framework/RobotKit(RKRobotDiscoveryAgent.o):
warning: /Users/Shared/TeamCity/bamboo-home/xml-data/build-dir/SDK-SDKI-ISR/build/ModuleCache/FBAX60G1D2VV/Darwin-MI6WZSG1PNOM.pcm: No such file or directory
[... cut ...]
while processing /Users/phsabo/Documents/SAS_Projects/MobilePathways/CodeSnaps_iOS/SpheroBlockly-iOS/RobotKit/RobotKit.framework/RobotKit(mock_spheroovmops.o):
warning: /Users/Shared/TeamCity/bamboo-home/xml-data/build-dir/SDK-SDKI-ISR/build/ModuleCache/3T3W1I53RCLX5/Darwin-MI6WZSG1PNOM.pcm: No such file or directory
while processing /Users/phsabo/Documents/SAS_Projects/MobilePathways/CodeSnaps_iOS/SpheroBlockly-iOS/RobotKit/RobotUIKit.framework/RobotUIKit(RUICalibrateOverlayView.o):
warning: /Users/Shared/TeamCity/bamboo-home/xml-data/build-dir/SDK-SDKI-ISR/build/ModuleCache/FBAX60G1D2VV/QuartzCore-3I7S0M4UEAD0U.pcm: No such file or directory
```

zenelk, 2016-10-05:
I'll get this tracked internally. In the mean time, is this causing any sort of adverse effects on your app, or just a lot of warnings?

pip8786, 2016-10-05:
No adverse effects that I can tell.

pip8786, 2016-11-02:
This is fixed in the beta branch.

closed 2016-11-02 by pip8786