sphero-inc/Sphero-iOS-SDK #51: RKResponseFactory is throwing an exception which is causing my app to crash

michaeldeitcher, 2016-10-03:
This is happening on a separate thread from my call to robot.send with a roll command. How do I keep it from crashing my app? Please help.

[RKResponseFactory responseFromRawPacket:withCommand:] - Bad Command ~ firmware mismatch? ( 4 ) for command <RKGetSkuCommand 0x02:0x3A : seq=0x05> - seq:0x05
2016-10-03 13:42:18.221077 Colorbot-iOS[667:125002] *** Terminating app due to uncaught exception 'NSRangeException', reason: '*** -[__NSArrayM insertObject:atIndex:]: index 17 beyond bounds [0 .. 15]'
*** First throw call stack:
(0x18fcf81c0 0x18e73055c 0x18fbd7ad4 0x10016ab08 0x100176200 0x100175fb8 0x100132c08 0x100c6525c 0x100c6521c 0x100c72b54 0x100c68ce4 0x100c730f0 0x100c74e6c 0x100c74bb8 0x18ed892c8 0x18ed88db4)
libc++abi.dylib: terminating with uncaught exception of type NSException

michaeldeitcher, 2016-10-04:
I've reduced my number of calls from 30 a second to 10 a second. It seems to be helping.

zenelk, 2016-10-05:
Hey michaeldeitcher, 
I'd love to look into this for you, but I'll need to know your SDK checkout first so I'm looking in the right spot.