sphero-inc/Sphero-iOS-SDK #52: Unable to get any response from Sphero

pip8786, 2016-10-05:
I'm trying to just read the odometer and I get this in the console:

```OBX:+[RKResponseFactory responseFromRawPacket:withCommand:] - Bad Command ~ firmware mismatch? ( 4 ) for command <RKReadOdometerCommand 0x02:0x75 : seq=0x01> - seq:0x01```

The response observer methods are never called for me in any case. 

Also right when the sphero connects I get errors like:
```
OBX:+[RKResponseFactory responseFromRawPacket:withCommand:] - Command ID: 4
OBX:+[RKResponseFactory responseFromRawPacket:withCommand:] - Command Device ID: 1
OBX:+[RKResponseFactory responseFromRawPacket:withCommand:] - Command Packet: <ffff0104 0001f9>
OBX:+[RKResponseFactory responseFromRawPacket:withCommand:] - Error Response 0x09 for command <RKJumpToMainAppCommand 0x01:0x04 : seq=0x00> - seq:0x00
```

I've made sure the sphero has the most up to date firmware, and the RobotKit is the latest from this repository.

zenelk, 2016-10-05:
Hey there,

For the first issue, depending on what version of Sphero you have, it might not actually have that command in firmware. I looked at the documentation we have and it looks like Sphero 2.0 / SPRK and up support it, but nothing with firmware version 1.xx. That error you posted is the firmware telling you that the command doesn't exist.

For the second, that is actually normal. What is happening is that when the robot connects, the SDK issues a "Jump to Main App" command to get it out of the bootloader (the slow light rainbow) mode so that we can use it. If it's not in the bootloader, it will return error code 0x09 saying that it's already in the main app. That error message should have no effect on your app.

As for not receiving the callbacks for the responses, what checkout of the SDK are you on?

pip8786, 2016-10-05:
@zenelk I'm using a Sphero 2.0, firmware 3.75 according to the Sphero app, and the SDK is the latest one from this site (downloading the master zip file). The issue I was trying to resolve by pinging the sphero with a command was that if the sphero connects to bluetooth while you're doing discovery, it will says it's connected but any commands we send to it won't work (including heading, sleep, drive, etc). I was hoping to essentially be able to tell that error state from a good connection and somehow resolve it.

zenelk, 2016-10-05:
Hmm. Okay. I'll get this looked at. Thanks for bearing with me.

gabriellet, 2016-10-12:
I also have this problem, and am using Sphero 2.0 and firmware 3.75. Even when not trying to read from the odometer I see a lot of `Bad Command ~ firmware mismatch?` messages in the console (For example when trying to send a drive instruction) and also the same errors when the Sphero connects.

pip8786, 2016-10-12:
@zenelk Any update on this? We're trying to release a product within a few weeks and I'm still struggling to keep a solid connection to the sphero through all the errors that are generated. Once it gets into a bad state it tends to be very difficult to get it back into an online and responsive state. The best way I know how is to go to the official Sphero app which seems to be able to reset it properly, then go back to my app and it works well. It would be nice to know what procedures the official app use to get connected that we may not be doing in our app?

gabriellet, 2016-10-20:
@pip8786 Have you made any progress on this? What are you developing in? Swift or ObjectiveC? I'm considering writing a packet and sending it manually to see if maybe the definitions in the SDK are out of date. Have you tried anything like that?

iamcgn, 2016-10-20:
@pip8786 we released an updated SDK under the beta branch. Please give that a shot and let me know if you still have any issues.

pip8786, 2016-11-02:
With the beta SDK I'm still getting firmware mismatch errors when just connecting to the robot, but I haven't tried sending it commands or reading sensors yet. Will come back to it when I get a chance.