sphero-inc/Sphero-iOS-SDK #53: Static Linking Errors and Warnings

gabriellet, 2016-10-12:
Hey, I've been finding lots of warnings and errors with incorrect static linking in the sample code. Does anyone know what's up with that?

[image]

pip8786, 2016-10-12:
This was reported here: https://github.com/orbotix/Sphero-iOS-SDK/issues/50

gabriellet, 2016-10-12:
Ah I see. Glad I'm not the only one who noticed it.

gabriellet, 2016-10-12:
It's also mentioned in #42 which is almost a year old now.

iamcgn, 2016-10-20:
This issue should be fixed in the latest beta branch. Please give that a shot and let me know if you're still seeing these warnings.

gabriellet, 2016-10-20:
@iamcgn Oh awesome, thanks! I'll let you know as soon was I try it out.