sphero-inc/Sphero-iOS-SDK #57: ANE Help

ckb2016, 2016-11-02:
Hi, I am trying to build an ane with the frameworks for an ActionScript mobile app. I can build the ane just fine. However, when I try to export a release build, I get this error:

Error occurred while packaging the application:

Undefined symbols for architecture armv7:

  "__ZNSt3__112basic_stringIcNS_11char_traitsIcEENS_9allocatorIcEEE6insertEmPKc", referenced from:

      __ZN7AstNode21new_UnaryOperatorNodeEPcPS_ in OvalCompiler(AstNode.o)

  "__ZNSt3__112basic_stringIcNS_11char_traitsIcEENS_9allocatorIcEEE7reserveEm", referenced from:

      __Z8toStringIiENSt3__112basic_stringIcNS0_11char_traitsIcEENS0_9allocatorIcEEEET_ in OvalCompiler(OvmCodeGenerator.o)

  "__ZNSt3__112basic_stringIcNS_11char_traitsIcEENS_9allocatorIcEEE6appendEPKc", referenced from:

      __ZN16OvmCodeGenerator20generateCode_idValueEP7AstNodeP10OvmContextRNSt3__16vectorI9OvmData_tNS4_9allocatorIS6_EEEE in OvalCompiler(OvmCodeGenerator.o)

      __ZN16OvmCodeGenerator28generateCode_declareVariableEP7AstNodeP10OvmContextRNSt3__16vectorI9OvmData_tNS4_9allocatorIS6_EEEE in OvalCompiler(OvmCodeGenerator.o)

Here is my platformoptions.xml:

<platform xmlns="http://ns.adobe.com/air/extension/19.0">
  <sdkVersion>4.6.0</sdkVersion>
  <linkerOptions>
    <option>-ios_version_min 6.0</option>
    <option>-framework Foundation</option>
    <option>-framework UIKit</option>
    <option>-framework CoreBluetooth</option>
    <option>-framework CoreGraphics</option>
    <option>-framework CoreMotion</option>
    <option>-framework ExternalAccessory</option>
    <option>-F [frameworks directory]</option>
    <option>-framework OvalCompiler</option>
    <option>-framework RobotKit</option>
    <option>-framework RobotUIKit</option>
  </linkerOptions>
</platform>

This is my extension.xml:

<extension xmlns="http://ns.adobe.com/air/extension/19.0">
  <id>RobotAneDemo</id>
  <versionNumber>1.0.0</versionNumber>
  <platforms>

    <platform name ="default">
      <applicationDeployment/>
    </platform>

    <platform name ="iPhone-ARM">
      <applicationDeployment>
        <nativeLibrary>libRobotAneDemo.a</nativeLibrary>
        <initializer>AirRobotExtInitializer</initializer>
      </applicationDeployment>
    </platform>
    
  </platforms>
</extension>

And this is my adt script:

adt -package -target ane RobotController.ane extension.xml -swc Interface.swc -platform iPhone-ARM -platformoptions platformoptions.xml -C ios/ . -platform default -C default/ .

I was wondering if anyone knew why this is failing. We can do apps in Objective-C or Swift, but we are more familiar with ActionScript. If we cannot get an ANE working, we can always do the apps entirely in Objective-C or Swift.

zenelk, 2016-11-02:
Hey there, is it possible that you are not able to use code that references C++ STL? I don't really know anything about actionscript or ANE, but it looks like you're just not able to see some STL symbols. Maybe you could include the C++ and STL dylibs in your platformoptions.xml to test this theory?

ckb2016, 2016-11-02:
Yes, that was the issue. I figured I was missing some dependency. Thank you.

closed 2016-11-02 by ckb2016