sphero-inc/Sphero-iOS-SDK #58: Determine which type of robot is connected

pip8786, 2016-11-02:
Using the SDK, is there a way to determine which type of robot you just connected to in the RobotStateNotification? Currently we do something like this:
```
r = notification.robot
if r.name() == "Sphero” {
    return RobotConfigs.sphero
} else if r.name().hasPrefix("BB-") {
    return RobotConfigs.bb8
} else if r.name().hasPrefix("SK-") {
    return RobotConfigs.spark
} else {
    return RobotConfigs.ollie
} 
```

Will this always work? Is there a better way?

Thanks,
Philippe

zenelk, 2016-11-02:
Hey there, for now this is the best way to do it. For robots that use Bluetooth LE, the name prefixes will always be consistent like those there. For Sphero, the actual advertised name could change if you modify it using commands. Here is a code snippet for how we identify robots for use in a module atop the SDK. Should help you out with this.
`
    if ([robot isKindOfClass:[RKRobotLE class]]) {
        if ([[robot name] hasPrefix:@"2B"]) {
            // Is an Ollie
        } else if ([[robot name] hasPrefix:@"BB"]) {
            // Is BB-8
        } else if ([[robot name] hasPrefix:@"1C"]) {
            // Is a Tencent WeBall
        } else if ([[robot name] hasPrefix:@"SK"]) {
            // Is a SPRK+
        }
    }
    else {
        // Is a Sphero
    }
`

closed 2016-11-02 by pip8786

lilliesAndRoses, 2017-01-03:
Should I therefore assume that the beta branch of this SDK supports BB8 alike SPRK+? And if I get SPRK working with this SDK, will I be able to use BB8 in the same way?

pip8786, 2017-01-03:
We were not able to get the BB8 working with the beta of the SDK. Only the SPRK+. Maybe that has changed since a couple months back.