sphero-inc/Sphero-iOS-SDK #59: Ollie's Set Heading Issues

pip8786, 2016-11-02:
Setting the heading for ollie will always make the angle come up short. This is true using the API in our app and within the Lightning Lab App. Is there a way to get around this? The heading setting seems to work well for all the other robots. It's important for our application that we are able to accurately set specific headings.

zenelk, 2016-11-02:
Hey there, I don't understand what you mean by "make the angle come up short". Could you explain a bit more so I know what to look for?

pip8786, 2016-11-02:
If I tell it to set heading to 90 degrees, move 0 velocity, it will turn but to about 75 degrees instead of 90, even with the backlight on. We're trying to separate the moving from the turning in our application to make it easier to time our movement for distance.