sphero-inc/Sphero-iOS-SDK #60: Does this SDK support BB8?

lilliesAndRoses, 2017-01-02:
Which all versions of sphero are supported by this SDK? 
Does the beta support SPRK and BB8? 
Why is this information not captured in the documentation?

pip8786, 2017-01-03:
We found that the BB8 was not supported in the beta, it works with the master version. The beta supports sprk+ though but not the BB8 so you have to pick.

lilliesAndRoses, 2017-01-03:
Thats sad! Is the beta branch integration with master planned in near future?

jamoore5, 2017-12-29:
Is there any documentation for the SPRK+ support? Should the HelloWorld sample work with SPRK+?

roberte3, 2017-12-29:
The hello world app should work the same as long as the ask supports that robot. (Haven’t tested it, though).

jamoore5, 2017-12-29:
There is a closed issue that says the beta branch supports for the SPRK+ robot.